RobTillaart/MAX31855_RT #1: Improvement Request

Banfotech, 2020-08-24:
Hello Rob!

Could be usefull to have the possibility to take the raw value out from the class.
Is possible to have the rawData keyword to have the raw value from the board?

Also, could be usefull to insert a fourth error status, "No Board", checking if all the bits of the message are == 0.
I'm doing some test but the actual code is not discriminating from no thermocouple and no board at all.

RobTillaart, 2020-08-24:
Hi,
Good ideas,

**No board**
Definitely useful as in real life broken wires / loose connector do happen. Mostly after some time.

I don't know if checking all bits are zero will work in all cases. I am thinking of broken wire and a floating pin. That would result in random bits so not all necessary zero. Using a pull up would make all bits 1 prevents a floating line, but if it would solve all or disrupts something else needs more thinking / investigation. Agree?

**raw data**
would not be difficult, something like line 99  
```cpp
uint32_t value = 0; 
``` 
should become a global var **raw** (within the class) and then add a 

```cpp
uint32_t getRawData() { return raw; };
```
The access function prevents overwriting the raw data from an application.

Please post a Pull Request for **raw data** it should be straightforward and I have little time this week.
(working on another lib)

Banfotech, 2020-08-24:
> needs more thinking / investigation. Agree?

For sure, at the moment i'm out of stock of boards, in a few days i should receive new one.

> Please post a Pull Request for raw data it should be straightforward and I have little time this week.

I think everything i've done it correctly

Pull request #2

pmarchini, 2020-08-24:
@RobTillaart Hey there,

I was working with @Banfotech in order to give him a little bit of help.
We were discussing about your library and found the issue above.

As you did i suggest him to try with a pull-up resistor on MOSI line in order to check if it's possible to have a line of 1's without issues to MAX31855 functionality.
ASAP we'll test and let you know, i've checked also adafruit library and found that even theirs is not able to understand if a board is connected or not and to me it would be great to add this feature.

In the meantime I'm going to make the PR you've requested

BR

RobTillaart, 2020-08-24:
👍

FabioBrondo, 2020-08-24:
| Bit set | Description |
|:----|:----|
| None | OK |
| 0 | thermocouple open circuit |
| 1 | thermocouple short to GND |
| 2 | thermocouple short to VCC |

RobTillaart, 2020-08-25:
This is now solved imho, the NO BOARD is named NO COMMUNICATION and the rawData is implemented too.

closed 2020-08-25 by RobTillaart