RobTillaart/MAX31855_RT #14: Error in MAX31855 when using HWSPI on STM32

Bibbbi, 2021-07-03:
When I use the original Max31855 library v0.2.4 the temperature display is not correct. (is to high)

[image]

I changed the source code as follows and the temeprature display was correct.

[image]

**MAX31855.cpp**
```
uint32_t MAX31855::_read(void)
{
  _rawData = 0;
  
  if (_hwSPI)
  {
    SPI.beginTransaction(SPISettings(16000000, MSBFIRST, SPI_MODE0));
    digitalWrite(_cs, LOW);
	
    for (uint8_t i = 0; i < 4; i++)
    {
      _rawData <<= 8;
      _rawData += SPI.transfer(0);
    }
    digitalWrite(_cs, HIGH);
    SPI.endTransaction();
  }
  else
  {
    digitalWrite(_cs, LOW);
    for (int8_t i = 31; i >= 0; i--)
    {
      _rawData <<= 1;
      digitalWrite(_sclk, LOW);
      // delayMicroseconds(1);  // DUE
      if ( digitalRead(_miso) ) _rawData++;
      digitalWrite(_sclk, HIGH);
      // delayMicroseconds(1);  // DUE
    }
    digitalWrite(_cs, HIGH);
  }
  return _rawData;
}
```
The CS signal must be between SPI.beginTransaction and SPI.endTransaction. Can anyone confirm this?

RobTillaart, 2021-07-04:
Seen the issue, might take some time to investigate (I do not have an STM32, so I have to check e.g ESP32)

It looks like a missing/extra pulse as the value seems x2

Can you confirm that the software SPI works?

Bibbbi, 2021-07-04:
yes,  Software SP works fine.

RobTillaart, 2021-07-04:
Good to hear SWSPI part is working.

<thinking out loud>

Technically if there is only one device it should be possible to connect the CS line to GND permanently. 
The SPI protocol should allow that.

looking at your log, I notice that the **RAW** value shows 26 bits (wrong value) and 25 bits (right value). 
It does not show the leading zero's - and it needs "byte separators"  [TODO]

The raw value (lower 11 bits of raw) is the temperature in steps of 0.0625°C  (1/16th)
As this value is shifted one position to the right, one will get a temperature that is exactly a factor 2 too high. 

So it appears that **SPI.beginTransaction(SPISettings(16000000, MSBFIRST, SPI_MODE0));** does trigger the sending of the first bit if the CS line is already LOW on the STM platform.

On AVR, UNO the library 0.2.4 works, but of course the low level HW SPI is different.
Your solution works too on a AVR, UNO.

Need to find a free ESP32 to test.

RobTillaart, 2021-07-04:
Created a develop branch / PR with your modifications plus some minor updates.

Tested with an ESP32 and an UNO at different SPI hardware speeds and works well.
Can you confirm the develop branch works with the STM32?

Thanks,
Rob

RobTillaart, 2021-07-04:
ESP32 run
[image]

Bibbbi, 2021-07-04:
I used your development branch
https://github.com/RobTillaart/MAX31855_RT/tree/develop

Tested with a STM32F405RGT6 at various SPI hardware speeds and works fine.
I can confirm that the development branch works with the STM32.

[image]

RobTillaart, 2021-07-05:
Thanks
Then I will merge later today.

commit 794867b2 (2021-07-05): Develop (#15)

* fix #14 CS for STM32
closed 2021-07-05 by RobTillaart