RobTillaart/MAX31855_RT #19: Random internal temperature & thermocouple readings

Z01NE, 2021-09-05:
I tested two modules with 3 boards, I get the same readings:
I added 10nF cap between T- and T+, and in one board, I cut the connection between T- and gnd
`20:47:19.425 -> Status:	STATUS_OK
20:47:19.425 -> RAW:	1100-1000-0100-1100-1000-0100-0101-0000
20:47:19.425 -> Internal:	-123.688
20:47:19.425 -> TMP:	-891.250
20:47:20.413 -> Status:	STATUS_OK
20:47:20.413 -> RAW:	1100-1000-0000-1000-1000-0100-0110-0000
20:47:20.413 -> Internal:	-123.625
20:47:20.413 -> TMP:	-895.500
20:47:21.405 -> Status:	STATUS_OK
20:47:21.405 -> RAW:	1100-0111-1111-0000-1000-0011-1101-0000
20:47:21.405 -> Internal:	-124.188
20:47:21.405 -> TMP:	-897.000
20:47:22.436 -> Status:	STATUS_OK
20:47:22.436 -> RAW:	1100-1000-0000-0100-1000-0011-1111-0000
20:47:22.436 -> Internal:	-124.063
20:47:22.436 -> TMP:	-895.750
20:47:23.455 -> Status:	STATUS_OK
20:47:23.455 -> RAW:	1100-1000-0001-1100-1000-0011-1000-0000
20:47:23.455 -> Internal:	-124.500
20:47:23.455 -> TMP:	-894.250
20:47:24.435 -> Status:	STATUS_OK
20:47:24.435 -> RAW:	1100-1010-1111-1100-1011-0011-0110-0000
20:47:24.435 -> Internal:	-76.625
20:47:24.435 -> TMP:	-848.250
20:47:25.440 -> Status:	STATUS_OK
20:47:25.440 -> RAW:	1100-1011-1111-1000-1100-0011-0001-0000
20:47:25.440 -> Internal:	-60.937
20:47:25.440 -> TMP:	-832.500
20:47:26.434 -> Status:	STATUS_OK
20:47:26.434 -> RAW:	1100-1100-1111-0000-1101-0010-1111-0000
20:47:26.434 -> Internal:	-45.062
20:47:26.434 -> TMP:	-817.000
20:47:27.432 -> Status:	STATUS_OK
20:47:27.432 -> RAW:	1100-1010-1111-1000-1011-0010-1110-0000
20:47:27.432 -> Internal:	-77.125
20:47:27.432 -> TMP:	-848.500
20:47:28.459 -> Status:	STATUS_OK
20:47:28.459 -> RAW:	1100-1011-0010-1000-1011-0010-1110-0000
20:47:28.459 -> Internal:	-77.125
`
Thank you

RobTillaart, 2021-09-05:
Strange readings.
Not enough information to draw any quick conclusion.

Z01NE, 2021-09-05:
Can you shed some light on where I should look?
Thanks

Z01NE, 2021-09-05:
It's the same with other libs.
I just want to know if possible, 
does a working chip report correct internal readings without a thermocouple connected?
Thanks

RobTillaart, 2021-09-06:
Some questions
- which boards tested?
- which pins?
- which example used?
- hardware SPI  or software SPI (did you test both?
- where did you buy the max31855 ?
- did you check soldering on the board?
- length of the wires used?
- do you use pull-ups resistors?
- if so which value?
- have you checked power voltage?  Stable?
-
A lot of questions and the answers provides me insight of how your test setup looks like. That might give some clue what causes the readings to fail

RobTillaart, 2021-09-06:
What are the expected temperatures (approx?)

RobTillaart, 2021-09-06:
```
                     TEMPERATURE             INTERNAL           STAT
20:47:19.425 -> RAW: 1100-1000-0100-11  00-  1000-0100-0101-   0000
20:47:20.413 -> RAW: 1100-1000-0000-10  00-  1000-0100-0110-   0000
20:47:21.405 -> RAW: 1100-0111-1111-00  00-  1000-0011-1101-   0000
20:47:22.436 -> RAW: 1100-1000-0000-01  00-  1000-0011-1111-   0000
20:47:23.455 -> RAW: 1100-1000-0001-11  00-  1000-0011-1000-   0000
20:47:24.435 -> RAW: 1100-1010-1111-11  00-  1011-0011-0110-   0000
20:47:25.440 -> RAW: 1100-1011-1111-10  00-  1100-0011-0001-   0000
20:47:26.434 -> RAW: 1100-1100-1111-00  00-  1101-0010-1111-   0000
20:47:27.432 -> RAW: 1100-1010-1111-10  00-  1011-0010-1110-   0000
20:47:28.459 -> RAW: 1100-1011-0010-10  00-  1011-0010-1110-   0000
```

There are some relative big jumps in the bit patterns  esp 5 -> 6  jumps of 10/15 degrees which is not expected.

Z01NE, 2021-09-06:
> which version of the library?

Current Master

> which boards tested?
which pins?
hardware SPI or software SPI (did you test both?

-nodeMcu 1.0 ESP 12E  (CLK D4 MISO D6 CS D4) 
-Arduino Yun (ATmega32u4)  (CLK 13 MISO 12 CS 7) 
-STM32f1103c8 BluePill (CLK PA5 MISO PA6 CS PA4) 
In SW SPI:
I tested more PIN configurations, I get "STATUS_OK" with random readings both internal and thermocouple.
In Hardware SPI:
I can't get it working, I get always "STATUS_NO_COMMUNICATION".

> where did you buy the max31855 ?

From Aliexpress two generic boards, a MAX chip with a cap between Vcc and Gnd.

> did you check soldering on the board?
length of the wires used?
do you use pull-ups resistors?
if so which value?
have you checked power voltage? Stable?

Setup in a breadboard with a level shifter for Arduino Yun
For other boards tested with breadboard and directly with jump wire of 12cm
I used a 1K and a 2K pull-up resistor

> have you checked power voltage? Stable?
From USB port, checked with a multimeter, it seems right.

>What are the expected temperatures (approx?)
Testing in room temperature 

I think I close and try to buy it from another source.

Thank you for your insights.

RobTillaart, 2021-09-06:
hardware SPI is tested by me on uno, and ESP32 (possibly NANO too), 
so looks like a different set - although your nodeMCU is that a ESP32?

software SPI should work on any platform in theory ..... (from famous last quotes)

I bought my max31855 from an ADAfruit reseller - 

pull up resistors sound OK.
Wire length sound OK

Do you have a link to the aliexpress thingie?

Z01NE, 2021-09-06:
nodeMCU v1.0 is an ESP8266 12E

> software SPI should work on any platform in theory ..... (from famous last quotes)

Good to know .

> I bought my max31855 from an ADAfruit reseller -

Better than having a headache 

> Do you have a link to the aliexpress thingie?

[https://www.aliexpress.com/item/1005002526054145.html](url)
This one have a short between T- and Gnd pin.

[https://www.aliexpress.com/item/32306974965.html](url)
This one sound ok.

closed 2021-09-07 by Z01NE

RobTillaart, 2021-09-10:
@Z01NE 
See that you closed the issue, is the cause understood?
Can you explain in short here for future reference?

Z01NE, 2021-09-10:
No, not really I suspect a faulty chip, I'm waiting for another one, I'll let you know when tested.

jpsminix, 2021-12-07:
I have exactly the same problem. All random readings from internal temp and thermocouple temp BUT ok with bitflags control.
NodeMCU ESP32 <SPI(sw) 8cm wire lenght> MAX31855 (clone from ebay and cut T- and GND)
I have oscilloscope to check SPI readings (software) and i see good comms. I have tested ok bit flags from erros from the thermocouple (SHORT to GND, SHORT to VCC, no thermocouple). When i see the bits from rawdata and check with oscilloscope, the data match.

ESP32 dont work for me with SPI HW on VSPI nor HSPI.

I have questions:
 - I see on the oscilloscope the CLK inverted (VCC at start) and on the datasheet i see its GND at start. Anyway it take good data when fall CLK.
 - I see on oscilloscope that freq is 2Mhz with sw, and i dont know if it is need to reach 5Mhz.
 - Without thermocouple connected, the internal temp still works?

Thank you for your time :)

RobTillaart, 2021-12-07:
@jpsminix  
I see only one explicit question **Without thermocouple connected, the internal temp still works?**

The answer is I never tried, as I have no hardware setup / time, I cannot test it.

From your description I understand that
1. software SPI is working for UNO 
2. software SPI is working for ESP32
3. hardware SPI is working for UNO
4. hardware SPI is failing for ESP32 (both HSPI and VSPI)

Can you confirm?

(it will be at least the weekend before I have time to dive into this.

RobTillaart, 2021-12-07:
@jpsminix 
you removed a post, this way the discussion makes no sense, please restore

sergiocntr, 2022-09-08:
hello everyone, just to thank Rob for this great library and z01ne for putting me on the right path  looking for the problem of my board (same) giving a short to ground error: with a cutter I removed the track near the chip and thing is resolved.  Strange I had to add an offset of +10 C.

RobTillaart, 2022-09-08:
@sergiocntr 
You're welcome,

> Strange I had to add an offset of +10 C.

Never heard before of such deviation (damage?), fortunately the library does support just offsets 😁

sergiocntr, 2022-09-09:
@RobTillaart maybe....  I just need to know a fireplace fumes temperature for now, it does not need to be very precise.( I hope that the difference remains constant as the temperature rises, for now I have checked it with ice) I'll do some tests with a DS18b20. In the lib Error handling was also very useful.👍

RobTillaart, 2022-09-09:
What temperature range are you talking about?
Fireplaces can be very hot I can imagine,

sergiocntr, 2022-09-09:
Seeing the colors on the stainless steel flue (you know that steel colors when exposed to high temperatures) I don't think it goes above 300 degrees Celsius in case it lets go too much.   the flame.

RobTillaart, 2022-09-09:
Given the range a thermocouple is the way to go.

You might have a look at https://learn.adafruit.com/adafruit-mlx90640-ir-thermal-camera?view=all
_This part will measure temperatures ranging from -40°C to 300°C with an accuracy of +- 2°C (in the 0-100°C range). With a maximum frame rate of 16 Hz (the theoretical limit is 32Hz but we were not able to practically achieve it), It's perfect for creating your own human detector or mini thermal camera._

sergiocntr, 2022-09-09:
Fine! Grazie. 👍

RobTillaart, 2022-09-09:
Or - https://learn.adafruit.com/using-melexis-mlx90614-non-contact-sensors