RobTillaart/MAX31855_RT #21: Random reads on temps

jpsminix, 2021-12-07:
I have exactly the same problem. All random readings from internal temp and thermocouple temp BUT ok with bitflags control.

[image]
[image]

NOTE: On this image, there is a flag, but when i say it give random temp data all flags is 0.

NodeMCU ESP32 <SPI(sw) 8cm wire lenght> MAX31855 (clone from ebay and cut T- and GND)
Arduino UNO <SPI(sw) 8cm wire lenght> MAX31855 (clone from ebay and cut T- and GND)

I have oscilloscope to check SPI readings (software) and i see good comms. I have tested ok bit flags from erros from the thermocouple (SHORT to GND, SHORT to VCC, no thermocouple). When i see the bits from rawdata and check with oscilloscope, the data match.

ESP32 dont work for me with SPI HW on VSPI nor HSPI.

Oh, i have tested with 3 clones from ebay, and 1 from aliexpress. All same problem.

I have questions:
 - I see on the oscilloscope the CLK inverted (VCC at start) and on the datasheet i see its GND at start. Anyway it take good data when fall CLK. 
 - I see on oscilloscope that freq is 2Mhz with sw, and i dont know if it is need to reach 5Mhz.
 - Without thermocouple connected, the internal temp still works?

Thank you for your time :)

RobTillaart, 2021-12-07:
@jpsminix
I see only one explicit question Without thermocouple connected, the internal temp still works?

The answer is I never tried, as I have no hardware setup / time, I cannot test it.

From your description I understand that

software SPI is working for UNO
software SPI is working for ESP32
hardware SPI is working for UNO
hardware SPI is failing for ESP32 (both HSPI and VSPI)
Can you confirm?

(it will be at least the weekend before I have time to dive into this.

jpsminix, 2021-12-07:
Hi, thank you to much for helping me

software SPI is working for UNO
software SPI is working for ESP32
hardware SPI is failing for UNO  (sorry for not write this before)
hardware SPI is failing for ESP32 (both HSPI and VSPI)

I have same setup on arduino for sw and hw SPI:
```
const int csPin   = 10;
const int clkPin  = 13;
const int dataPin = 12;

uint32_t start, stop;

MAX31855 tc(csPin);
//MAX31855 tc(clkPin, csPin, dataPin);  // sw SPI
```

And only change the last line to this to make sw SPI working:
```
//MAX31855 tc(csPin);
MAX31855 tc(clkPin, csPin, dataPin);  // sw SPI
```

I put some log with hw SPI:
16:51:59.675 -> time:		480
16:51:59.675 -> stat:		129
16:51:59.675 -> raw:		1111 1111 1111 1111 1111 1111 1111 1111
16:51:59.721 -> internal:	0.000
16:51:59.721 -> temperature:	nan
16:52:00.691 -> 
16:52:00.691 -> time:		484
16:52:00.691 -> stat:		129
16:52:00.691 -> raw:		1111 1111 1111 1111 1111 1111 1111 1111
16:52:00.691 -> internal:	0.000
16:52:00.691 -> temperature:	nan
16:52:01.705 -> 

I put some log with sw SPI:
16:52:42.680 -> time:		508
16:52:42.680 -> stat:		0
16:52:42.680 -> raw:		0111 1000 0011 1000 1001 1101 1101 0000
16:52:42.726 -> internal:	-98.188
16:52:42.726 -> temperature:	1923.500
16:52:43.699 -> 
16:52:43.699 -> time:		508
16:52:43.699 -> stat:		0
16:52:43.699 -> raw:		0111 1100 0011 1100 1001 1101 1100 0000
16:52:43.699 -> internal:	-98.250
16:52:43.699 -> temperature:	1987.750

I have a pull up resistor ~2k between 5V and DO (MISO).

MAX31855 is connected to 5V (has a regulator to 3.3v the board).

This is a capture of oscilloscope of Arduino UNO, sw SPI, (pull up resistor 2k only on MISO) and MAX31855 (with LDO to 3.3v):
[image]

RobTillaart, 2021-12-07:
OK,  which time zone are you in EU  or USA?  
Helps to know the moments we are both online, I am from the Netherlands (EU).
 
the first raw: 1111 1111 1111 1111 1111 1111 1111 1111   is definitely wrong

The SW SPI values are correct ? (just asking as 1923° seems quite high and -98° seems very low.
```
16:52:42.680 -> raw: 0111 1000 0011 1000 1001 1101 1101 0000
16:52:42.726 -> internal: -98.188
16:52:42.726 -> temperature: 1923.500
```

jpsminix, 2021-12-07:
I am from Spain (EU).

The first raw it is beacuse hw SPI its not working (i dont see anything on the pins, CS/CLK/DO).

The SW SPI the values ar not correct, the are completely random. Internal temp can jump from -98 to 127 and then go to -127. Same with thermocouple. BUT the flags bits are correct. If i short the pins near GND to T- and VCC to T+, the flags are correct.

There is a big posibility that 4 MAX31855 that i have with same results maybe are faulty (all clones). But i would like to make it work with hw SPI.. maybe sw SPI is slow... (i dont think so because with sw SPI and ESP32 its pretty fast)... but i dont know what to check.

RobTillaart, 2021-12-07:
Have you tried e.g. Adafruits library? 
Might help to check if it is in the hardware / wiring  or in the software.
If I find some time tomorrow I will setup a test environment, I hope I have all the HW needed.

jpsminix, 2021-12-08:
I just installed now the adafruit MAX31855 v1.3 and configured for HW SPI with CS=10 with all same wires (nothing changed)
```
// Example creating a thermocouple instance with software SPI on any three
// digital IO pins.
//#define MAXDO   3
//#define MAXCS   4
//#define MAXCLK  5

// initialize the Thermocouple
//Adafruit_MAX31855 thermocouple(MAXCLK, MAXCS, MAXDO);

// Example creating a thermocouple instance with hardware SPI
// on a given CS pin.
#define MAXCS   10
Adafruit_MAX31855 thermocouple(MAXCS);

// Example creating a thermocouple instance with hardware SPI
// on SPI1 using specified CS pin.
//#define MAXCS   10
//Adafruit_MAX31855 thermocouple(MAXCS, SPI1);
```

The logs:
12:18:31.024 -> Internal Temp = 127.94
12:18:31.024 -> C = 639.00
12:18:31.992 -> Internal Temp = 127.94
12:18:32.038 -> C = 639.00
12:18:33.009 -> Internal Temp = 127.94
12:18:33.055 -> C = 642.25
12:18:34.024 -> Internal Temp = 127.94
12:18:34.024 -> C = 580.50
12:18:34.997 -> Internal Temp = 127.94
12:18:35.043 -> C = 646.50
12:18:36.013 -> Internal Temp = 127.94
12:18:36.059 -> C = 646.50

The oscilloscope view.. this confuse me a little (so much space on end of DO)
[image]

This is the same situation as before but with short T- to GND.
[image]

I think internal temp is broken on all my chips and its impossible to calculate the external temp without the reference of internal.

Thank you so much!

RobTillaart, 2021-12-08:
Is it correct that I to see that the clock is 5V and the dataline is 3V3 ?

RobTillaart, 2021-12-08:
> I think internal temp is broken on all my chips and its impossible to calculate the external temp without the reference of internal.

I need to dive into the datasheet / code to confirm that

RobTillaart, 2021-12-08:
Maybe your sensor batch does not work at full frequency. I've heard such stories for other sensors before, that 2nd choice batches are sold cheaply. 

(thinking out loud)
In my library you can patch the software timing by adding delayMicroseconds 
search for ```     // delayMicroseconds(1);  // DUE  ```
and in a similar way you can patch the hardware handshake too.  try e.g. 10 us per pulse. 
Absolutely no guarantee it will work, at least it could be tested in a few minutes if the sensor works on lower frequency.

RobTillaart, 2021-12-08:
The gap in the scope picture might be an artefact of the library used. Then you should see it after every data exchange.

jpsminix, 2021-12-08:
> Is it correct that I to see that the clock is 5V and the dataline is 3V3 ?

Yes, you are correct but i dont know why, the pull up resistor is on the pin of the DO.

[image]
.
Another difference is that CLK line start from LOW with adafruit (hw SPI)

jpsminix, 2021-12-08:
> 

Im gonna test it. I will update with news.

jpsminix, 2021-12-08:
I have changed the file MAX31855.cpp with this modification:
```
  else  // Software SPI
  {
    digitalWrite(_select, LOW);
    for (int8_t i = 31; i >= 0; i--)
    {
      _rawData <<= 1;
      digitalWrite(_clock, LOW);
      delayMicroseconds(10);   // DUE MODIFIED
      if ( digitalRead(_miso) ) _rawData++;
      digitalWrite(_clock, HIGH);
      delayMicroseconds(10);  // DUE MODIFIED
    }
    digitalWrite(_select, HIGH);
  }
```
With this modification i see that the Freq with sw SPI has changed from ~70khz to ~32Khz with same result.
Data log is a little more stable on external temp but internal temp is off:
14:27:32.386 -> time:		1056
14:27:32.386 -> stat:		0
14:27:32.386 -> raw:		0011 0100 0100 0000 1000 0000 0000 0000
14:27:32.386 -> internal:	-128.000
14:27:32.386 -> temperature:	836.000
14:27:33.360 -> 
14:27:33.360 -> time:		1056
14:27:33.360 -> stat:		0
14:27:33.360 -> raw:		0011 0100 0010 1000 1000 0000 0000 0000
14:27:33.360 -> internal:	-128.000
14:27:33.360 -> temperature:	834.500

jpsminix, 2021-12-08:
My ambiental temp is about 19~24º and all test is on ambiental temp.

RobTillaart, 2021-12-08:
Found a problem when testing with UNO. 
I noticed that the constructor that sets the hardware flag does not get executed (anymore?). 
When moving the code from the constructor to begin HW SPI starts to work again at least on UNO. 

### todo
- check if ESP32 HW has the same problem. 
- fix the code

RobTillaart, 2021-12-08:
- ESP32 shows same problem
- see on the scope that the ESP32 definitely needs pull-up resistors on the clock, and data line.

Fix needs a rewrite of the **constructor** and **begin()** function and will be a breaking change.
Might take some days to fix.

RobTillaart, 2021-12-08:
@jpsminix 
Created a branch https://github.com/RobTillaart/MAX31855_RT/tree/fix21 that should solve the issue.
Can you please test if this works with UNO SW HW and ESP32 SW and HW?

Note: In your sketch you need to move the parameters from the **constructor** to the **begin()**.

RobTillaart, 2021-12-08:
ESP32 via VSPI  (240MHz)
```
time:		15
stat:		0
raw:		0000 0001 0100 1000 0001 0011 1101 0000
internal:	19.812
temperature:	20.500
```
note that SW SPI is almost equally fast 17 vs 15 us. - that is ~2 bits / usec

jpsminix, 2021-12-08:
> @jpsminix Created a branch https://github.com/RobTillaart/MAX31855_RT/tree/fix21 that should solve the issue. Can you please test if this works with UNO SW HW and ESP32 SW and HW?
> 
> Note: In your sketch you need to move the parameters from the **constructor** to the **begin()**.
> 
> update: unit tests needs investigation update 2: unit tests fixed, pull request is ready for merging.

I have tested Arduino UNO HW and ok. Tomorrow i will test the rest:
 - ESP32 HW and SW
 - Arduino UNO SW again.

RobTillaart, 2021-12-09:
update on an open question:

> > I think internal temp is broken on all my chips and its impossible to calculate the external temp without the reference of internal.
>
> I need to dive into the datasheet / code to confirm that

From datasheet:

_The device senses and corrects for the changes in
the reference junction temperature with cold-junction
compensation. It does this by first measuring its internal
die temperature, which should be held at the same temperature
as the reference junction. It then measures the
voltage from the thermocouple’s output at the reference
junction and converts this to the noncompensated thermocouple
temperature value. This value is then **added**
to the device’s die temperature to calculate the thermocouple’s
“hot junction” temperature. Note that the “hot
junction” temperature can be lower than the cold junction
(or reference junction) temperature._

So yes, the internal temperature is needed as a reference. What I understand from the above is that the external sensor follows the internal sensor (as the external is added). So in theory the two temperatures should follow same pattern ?
However if internal is broken anything is possible in practice.

jpsminix, 2021-12-09:
Sorry, bad wiring on esp32 testing. I have corrected the pinout wiring and ESP32 with sw SPI -> OK. Im gonna test ESP32 hw SPI now.

RobTillaart, 2021-12-09:
> Sorry, bad wiring on esp32 testing. I have corrected the pinout wiring and ESP32 with sw SPI -> OK. Im gonna test ESP32 hw SPI now.

happens to me daily :)

jpsminix, 2021-12-09:
No data with ESP 32 hw SPI and 2 resistor pull up (~2k) on CLK and DO lines. I dont see anything on the scope. Log output:
17:18:30.257 -> time:		51
17:18:30.257 -> stat:		0
17:18:30.257 -> raw:		0000 0000 0000 0000 0000 0000 0000 0000
17:18:30.303 -> internal:	0.000
17:18:30.303 -> temperature:	0.000
17:18:31.272 -> 
17:18:31.272 -> time:		51
17:18:31.272 -> stat:		0
17:18:31.272 -> raw:		0000 0000 0000 0000 0000 0000 0000 0000
17:18:31.272 -> internal:	0.000
17:18:31.272 -> temperature:	0.000

jpsminix, 2021-12-09:
Wops, i dont have choose demo VPSI. i am gonna check.

jpsminix, 2021-12-09:
Yep, that was de VPSI command. My bad.

Ok ESP32 hw SPI working (with and without resistor pullup). Logs:
17:23:49.640 -> time:		16
17:23:49.640 -> stat:		0
17:23:49.640 -> raw:		0101 0100 1001 1000 1011 0011 1101 0000
17:23:49.640 -> internal:	-76.188
17:23:49.640 -> temperature:	1353.500
17:23:50.660 -> 
17:23:50.660 -> time:		16
17:23:50.660 -> stat:		0
17:23:50.660 -> raw:		0100 1111 1111 1000 1011 0110 0001 0000
17:23:50.660 -> internal:	-73.938
17:23:50.660 -> temperature:	1279.500

The scope its a little strange, but works:
[image]

jpsminix, 2021-12-09:
> ESP32 via VSPI (240MHz)
> 
> ```
> time:		15
> stat:		0
> raw:		0000 0001 0100 1000 0001 0011 1101 0000
> internal:	19.812
> temperature:	20.500
> ```
> 
> note that SW SPI is almost equally fast 17 vs 15 us. - that is ~2 bits / usec
> 
> good moment to stop :)

If you disconnect the thermocouple wiring, is the MAX31855 still gives you the readings of internal temp?

RobTillaart, 2021-12-09:
it looks like you got it working!

> If you disconnect the thermocouple wiring, is the MAX31855 still gives you the readings of internal temp?

I will try, will take few minutes

RobTillaart, 2021-12-09:
I removed the wire and immediately 
```
time:           15
stat:           1
raw:            0111 1111 1111 1101 0001 0111 0001 0001
internal:       0.000
temperature:    0.000
```

So it looks like the raw data is not completely "crap" 
I need to check if the read "returns" prematurely.

RobTillaart, 2021-12-09:
made a patch in the .cpp and it shows a nice internal and a faulty external  ==> bit 18 -31 are all 1

```
time:		16
stat:		1
raw:		0111 1111 1111 1101 0001 0011 1011 0001
internal:	19.687
temperature:	2047.750
```

RobTillaart, 2021-12-09:
(thinkin out loud)
That implies including this patch in the Pull Request gives better information in case of an STATUS_OPEN_CIRCUIT  (1) error.
Agree?

RobTillaart, 2021-12-09:
@jpsminix 
I removed the "premature" return from **read()** so it now it will always fill internal and temperature from the raw data.
Can you verify?

RobTillaart, 2021-12-09:
> 
> The scope its a little strange, but works (15.6Mhz): [image]

looks similar to what I saw on my scope yesterday.  The ESP32 the SW SPI had no time between the bytes, while HW SPI had faster bursts like your picture shows. So hardware SPI is in theory maybe twice as fast, but the overhead of transaction exists. In practice that means pulses are shorter and probably not a nice square ware, but more a "rounded shark fin". So for stability SW SPI should be preferred.

jpsminix, 2021-12-09:
> (thinkin out loud) That implies including this patch in the Pull Request gives better information in case of an STATUS_OPEN_CIRCUIT (1) error. Agree?

Complety agree!

jpsminix, 2021-12-09:
> @jpsminix I removed the "premature" return from **read()** so it now it will always fill internal and temperature from the raw data. Can you verify?

I will try on a while. I will tell you.

jpsminix, 2021-12-09:
Ok, its working (my chip is giving random numbers but works your code).

jpsminix, 2021-12-09:
> looks similar to what I saw on my scope yesterday. The ESP32 the SW SPI had no time between the bytes, while HW SPI had faster bursts like your picture shows. So hardware SPI is in theory maybe [...]

Correct!

RobTillaart, 2021-12-09:
I'm going to merge the Pull Request, thank  you for the issue and your testing

RobTillaart, 2021-12-09:
Issue is automatically closed by PR, so if there are new problems please open a new issue (or reopen if related)

closed 2021-12-09 by RobTillaart

RobTillaart, 2021-12-09:
@jpsminix 
I am thinking about adding a configurable delay in the SW SPI loop so the timing can be adjusted.

```cpp
  else  // Software SPI
  {
    digitalWrite(_select, LOW);
    for (int8_t i = 31; i >= 0; i--)
    {
      _rawData <<= 1;
      digitalWrite(_clock, LOW);
      if (del > 0 )  delayMicroseconds(del / 2 );         <<<<<<<<<
      if ( digitalRead(_miso) ) _rawData++;
      digitalWrite(_clock, HIGH);
       if (del > 0 ) delayMicroseconds(del - del / 2 );     <<<<<<<<<<<
    }
    digitalWrite(_select, HIGH);
  }
```
and then add  **void setSWSPIdelay(uint16_t del)**  and a **uint16_t getSWSPIdelay()**

this would allow adjusting timing when needed e.g. when long wires are used.

commit dbf9672f (2021-12-08): fix #21 move constructoer code to begin() - breaking change
commit 195afcde (2021-12-09): fix #21 move constructor code to begin() - breaking change (#22)

* fix #21 breaking change
* move constructor code to begin()
* fix examples
* removed "premature" return from read() so internal and temperature are always converted from raw data.