RobTillaart/MAX31855_RT #23: Add setSWSPIdelay()

RobTillaart, 2021-12-09:
- for tuning SW SPI signal timing e.g. long wires.
-

RobTillaart, 2021-12-09:
like this
```cpp
  else  // Software SPI
  {
    digitalWrite(_select, LOW);
    for (int8_t i = 31; i >= 0; i--)
    {
      _rawData <<= 1;
      digitalWrite(_clock, LOW);
      if (swdel > 0 )  delayMicroseconds(swdel / 2 );         <<<<<<<<<
      if ( digitalRead(_miso) ) _rawData++;
      digitalWrite(_clock, HIGH);
      if (swdel > 0 ) delayMicroseconds(swdel - swdel / 2 );     <<<<<<<<<<<
    }
    digitalWrite(_select, HIGH);
  }
```
and then add  **void setSWSPIdelay(uint16_t swdel)**  and a **uint16_t getSWSPIdelay()**

this would allow adjusting timing when needed e.g. when long wires are used.

jpsminix, 2021-12-09:
It would be nice to adjust. But i dont understand the second if (swdel > 0)... "swdel - swdel / 2" == 0/2 ??