RobTillaart/MAX31855_RT #27: Would this work on ESP32-S3?

AncientTree, 2023-11-05:
I have no idea about link pin between ESP32-S3 and MAX31855. ESP32-S3 pin function map is display below. If I ref this project for ESP32-S3, will `setGPIOpins()` work well like that on ESP32 ?

My pin defined like this: [https://github.com/Zanduino
MAX31855](https://github.com/Zanduino/MAX31855)
```
const uint32_t SERIAL_SPEED     = 115200; ///< Set the baud rate for Serial I/O
const uint8_t  SPI_CHIP_SELECT  =     10; ///< Chip-Select PIN for SPI
const uint8_t  SPI_MISO         =     37; ///< Master-In, Slave-Out PIN for SPI
const uint8_t  SPI_SYSTEM_CLOCK =     36; ///< System Clock PIN for SPI
```
Mentioned github project run well on Arduino uno R3, But the temperature always reported zero on ESP32-S3. :< Is there some guidance? Thanks!
```
Ambient Temperature is 0.000°C
Probe Temperature is   0.000°C

Ambient Temperature is 0.000°C
Probe Temperature is   0.000°C
```

[image]

RobTillaart, 2023-11-05:
Hi,
Thanks for the issue, might take a few days to dive into it . Point of attention is to have pull up resistors on the data and click lines. 
How do you power the device 3V3 or 5V?

AncientTree, 2023-11-06:
Thank you very much for attention and replying. Someone said that pull up resistors is unnecessary to SPI. The program run well on Arduino UNO R3 (which have build in pull up resistors) should run well on ESP32-S3 too, with some change in pin define of course. I use 5V on ESP32-S3.

1) What confuse me is that ESP32-S3 pin function map declared two `FSPID` 11 and 35, two `FSPICLk` 12 and 36, two `FSPIQ` 13 and 37 and so on. As [ESP32-S3 Technical Reference Manual 30.5.2 Introduction to FSPI bus and SPI3 Bus Signals](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#page=1078) show that    `FSPIQ` similar to `MISO` on UNO R3, but `FSPIQ` have pin 13 and 37. Should I wiring MAX31855-SO to ESP32-S3 13 or 37 ? MAX31855-SCK, MAX31855-CS same problem.

2) You do a nice work that tell people MISO and CLOCKPIN is fixed, MOSI _not used_, Chip-Select _can be others too_. This is helpful for newbies. If it possible I hope that the table expand two column with `ESP32-S3 FSPI` and `ESP32-S3 SUBSPI`. And for soft SPI using, hope setGPIOpins() expanded to suite ESP32-S3 too. 

Sorry to disturb and Thanks for your outstanding work 👍.

RobTillaart, 2023-11-06:
> Sorry to disturb

No problem, just very busy

RobTillaart, 2023-11-12:
@AncientTree 

Have you tested the library with the software SPI?

AncientTree, 2023-11-12:
I can't verify that the code you show could work or not, because I have no idea about Espressif IDF. I had to make do with the hardware SPI wiring method I found. I finally found a usable wiring like this:

[image]

And my code like this, Arduino-IDE with library [https://github.com/Zanduino/MAX31855](https://github.com/Zanduino/MAX31855):

```
/*! @file Demo.ino

@section Demo_intro_section Description

Example program for measuring temperatures using the MAX31855 Cold-Junction Compensated Thermocouple-to-digital
converter.\n\n
The MAX31855 datasheet is located at https://datasheets.maximintegrated.com/en/ds/MAX31855.pdf and describes the
relatively simple interface.  The MAX31855 uses SPI to communicate and only outputs data. It writes a sentence of
32 bits which include devices failure information and the ambient temperature as well as the thermocouple probe 
temperature in degrees Celsius.\n\n

The most recent version of the MAX31855 library is available at https://github.com/SV-Zanshin/MAX31855 and the
documentation of the library as well as example programs are described in the project's wiki pages located at
https://github.com/SV-Zanshin/MAX31855/wiki \n\n

This example is simple in that it displays the ambient and probe temperatures every 5 seconds and flags any errors 
that the device raises.

@section Demolicense __**GNU General Public License v3.0**__

This program is free software: you can redistribute it and/or modify it under the terms of the GNU General
Public License as published by the Free Software Foundation, either version 3 of the License, or (at your
option) any later version. This program is distributed in the hope that it will be useful, but WITHOUT ANY
WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details. You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.

@section Demoauthor Author

Written by Arnd\@SV-Zanshin

@section Demoversions Changelog

Version | Date       | Developer                     | Comments
------- | ---------- | ----------------------------- | ---------------------------------------------------
1.0.1   | 2019-02-02 | https://github.com/SV-Zanshin | Issue #4 - Convert program comments and style to Doxygen
1.0.0   | 2017-08-12 | https://github.com/SV-Zanshin | Finished, first release
1.0.0a  | 2017-08-11 | https://github.com/SV-Zanshin | Initial coding
*/
#include <MAX31855.h> // Include MAX31855 Sensor library
/*******************************************************************************************************************
** Declare all program constants                                                                                  **
*******************************************************************************************************************/
const uint32_t SERIAL_SPEED     = 115200; ///< Set the baud rate for Serial I/O
const uint8_t  SPI_CHIP_SELECT  =     8; ///< Chip-Select PIN for SPI
const uint8_t  SPI_MISO         =     13; ///< Master-In, Slave-Out PIN for SPI
const uint8_t  SPI_SYSTEM_CLOCK =     12; ///< System Clock PIN for SPI

/*******************************************************************************************************************
** Declare global variables and instantiate classes                                                               **
*******************************************************************************************************************/
MAX31855_Class MAX31855; ///< Create an instance of MAX31855

/***************************************************************************************************************//*!
* @brief    Arduino method called once at startup to initialize the system
* @details  This is an Arduino IDE method which is called first upon boot or restart. It is only called one time
*           and then control goes to the main "loop()" method, from which control never returns
* @return   void
*******************************************************************************************************************/
void setup()
{
  Serial.begin(SERIAL_SPEED);
  #ifdef  __AVR_ATmega32U4__  // If this is a 32U4 processor, then wait 3 seconds for the interface to initialize
[... 20 lines cut ...]
*******************************************************************************************************************/
void loop()
{
  int32_t ambientTemperature = MAX31855.readAmbient(); // retrieve MAX31855 die ambient temperature
  int32_t probeTemperature   = MAX31855.readProbe();   // retrieve thermocouple probe temp
  uint8_t faultCode          = MAX31855.fault();       // retrieve any error codes
  if ( faultCode )                                     // Display error code if present
  {
    Serial.print("Fault code ");
    Serial.print(faultCode);
    Serial.println(" returned.");
  }
  else
  {
    Serial.print("Ambient Temperature is ");
    Serial.print((float)ambientTemperature/1000,3);
    Serial.println("\xC2\xB0""C");
    Serial.print("Probe Temperature is   ");
    Serial.print((float)probeTemperature/1000,3);
    Serial.println("\xC2\xB0""C\n");
  } // of if-then-else an error occurred
  delay(2000);
} // of method loop()

```

RobTillaart, 2023-11-23:
@AncientTree 
OK Thank you for the update, appreciated.

if there are no more questions you may close the ossue.

AncientTree, 2023-11-24:
Thank you for attention and my problem resolved perfectly.

closed 2023-11-24 by AncientTree

AncientTree, 2023-11-30:
I'm happy to tell you that your library and the software SPI sample (`examples\max31855_swSPI\max31855_swSPI.ino`) works perfectly with the ESP32-S3 and I can set any pin as DO, CLK. My judgment was previously clouded by a few broken Dupont wires(silly me). Now you may be able to add the ESP32-S3 to your compatibility list.